EyeTrackVR/EyeTrackVR #95: On installation not working on Windows

Abhinav0002, 2023-10-11:
Downloaded the .exe file from github, but only a blank cmd window is displayed on running the application and 

No settings file found, using base settings displayed in cmd

RedHawk989, 2023-10-11:
I need more information. What .exe, did it ever work and what app version?

Abhinav0002, 2023-10-11:
The eyetrack setup.exe file, and version is 0.1.8.1

RedHawk989, 2023-10-11:
Can you launch the app through a command prompt? (open a command prompt in the same folder as the exe and then type the exe name. Ex: eyetrackapp.exe

You should get an output. Paste it here.

Abhinav0002, 2023-10-11:
> Can you launch the app through a command prompt? (open a command prompt in the same folder as the exe and then type the exe name. Ex: eyetrackapp.exe
> 
> You should get an output. Paste it here.

I tried to launch it through the cmd and I am getting the same thing:

No settings file found, using base settings and after that nothing loads

Abhinav0002, 2023-10-13:
Nothing else is displayed on the cmd 
[image]

RedHawk989, 2023-10-13:
Have you tried just running from source? This is a very unusual error.

RedHawk989, 2024-02-12:
Issue is stale. Closing.

closed 2024-02-12 by RedHawk989