EyeTrackVR/EyeTrackVR #109: Corrupted config causes crash

Luois45, 2024-04-09:
### PROBLEM DESCRIPTION
The application crashes on startup if the config gets currupted. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/RedHawk989/EyeTrackVR/repo-tools/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/RedHawk989/EyeTrackVR/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/RedHawk989/EyeTrackVR/issues)
- [x] Searched the problem in [discussions](https://github.com/RedHawk989/EyeTrackVR/discussions)
- [x] Searched the problem in the [docs](https://redhawk989.github.io/EyeTrackVR/FAQ)
- [x] Searched the problem in the [chat](https://redhawk989.github.io/EyeTrackVR/contact/)
- [x] Device used (e.g., ESP32): -
- [x] EyeTrackVR binary firmware version number used: v0.1.8.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: -
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here: -

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here: -

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here: -

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here: -

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

[eyetrack_settings.json](https://github.com/EyeTrackVR/EyeTrackVR/files/14924292/eyetrack_settings.json)
Replace the config with the corrupted config I had. It got corrupted due to the perfect timing of a bluescreen.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Showing an error. (And maybe asking the user if he wants to get the config deleted.)

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

-

**(Please, remember to close the issue when the problem has been addressed)**

RedHawk989, 2024-04-10:
This is a very old bug, that is addressed in our upcoming v2 app https://github.com/EyeTrackVR/EyeTrackVR/tree/v2.0-beta-feature-branch

For now, delete the eyetrack_settings.json and it will be fixed.

Luois45, 2024-04-10:
> This is a very old bug, that is addressed in our upcoming v2 app https://github.com/EyeTrackVR/EyeTrackVR/tree/v2.0-beta-feature-branch
> 
> For now, delete the eyetrack_settings.json and it will be fixed.

Okay. I'll close this issue then. The problem is that I got bluescreened by EyeTrackVR 2 Times till now with each one resulting in a broken config.

closed 2024-04-10 by Luois45