EyeTrackVR/EyeTrackVR #111: Exiting EyeTrackApp but doesn't actually exit

darkfader, 2024-06-10:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
When trying to close the program, it just says "Exiting EyeTrackApp"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/RedHawk989/EyeTrackVR/repo-tools/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/RedHawk989/EyeTrackVR/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/RedHawk989/EyeTrackVR/issues)
- [x] Searched the problem in [discussions](https://github.com/RedHawk989/EyeTrackVR/discussions)
- [x] Searched the problem in the [docs](https://redhawk989.github.io/EyeTrackVR/FAQ)
- [x] Searched the problem in the [chat](https://redhawk989.github.io/EyeTrackVR/contact/)
- [x] Device used (e.g., ESP32): ESP32S3
- [X] EyeTrackVR binary firmware version number used: latest
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Press close button of app.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

The app should exit :3

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

RedHawk989, 2024-06-10:
> Steps to reproduce the behavior: Press close button of app.

See the problem is I've known about this bug but it isn't reproducible with a specific known set of conditions (it just happens depending on the state of the app) which makes it very hard to track the issue down for now. It's pretty low on the priority list as you can just x out of the terminal, but ill keep it in mind if I ever have the time to fix it.

RedHawk989, 2024-06-10:
Also can you provide the *specific* version number? "latest" isn't as clear as one would think.

darkfader, 2024-06-16:
I think the version is not super-relevant, but: commit cb0c618ce9a64d4fb1165ab384391eb66cbe15b0 (HEAD -> v2.0-beta-feature-branch, origin/v2.0-beta-feature-branch)

The work-around I used:
`            print("\033[94m[INFO] Exiting EyeTrackApp\033[0m")
            os._exit(0)    # <- this
`

RedHawk989, 2024-07-04:
Adding some notes here, it seems to be an issue with LEAP's multithreading with the threads not being shut down properly. Making LEAP not use the threaded queues fixes this issue, however, results in lower performance. When it is added back, I will be trying to make sure it can properly stop the threads.

RedHawk989, 2024-07-05:
implemented your fix in the meantime, marking as closed.

closed 2024-07-05 by RedHawk989