EyeTrackVR/EyeTrackVR #114: Build failure when running make pyinstaller inside docker image on v2.0-beta-feature-branch

rain-1, 2024-07-03:
Hello.

I tried to build v2.0-beta-feature-branch from source. I build the docker image and then entered the docker image and ran make install then make pyinstaller. This resulted in the following errors:

```
12530 INFO: Looking for eggs
12531 INFO: Python library not among binary dependencies. Performing additional search...
Traceback (most recent call last):
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/bin/pyinstaller", line 8, in <module>
    sys.exit(_console_script_run())
             ^^^^^^^^^^^^^^^^^^^^^
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/__main__.py", line 194, in _console_script_run
    run()
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/__main__.py", line 180, in run
    run_build(pyi_config, spec_file, **vars(args))
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/__main__.py", line 61, in run_build
    PyInstaller.building.build_main.main(pyi_config, spec_file, **kwargs)
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/building/build_main.py", line 1019, in main
    build(specfile, distpath, workpath, clean_build)
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/building/build_main.py", line 944, in build
    exec(code, spec_namespace)
  File "EyeTrackApp/eyetrackapp.spec", line 9, in <module>
    a = Analysis(
        ^^^^^^^^^
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/building/build_main.py", line 429, in __init__
    self.__postinit__()
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/building/datastruct.py", line 184, in __postinit__
    self.assemble()
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/building/build_main.py", line 764, in assemble
    self._check_python_library(self.binaries)
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/building/build_main.py", line 834, in _check_python_library
    python_lib = bindepend.get_python_library_path()
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/appuser/.cache/pypoetry/virtualenvs/eyetrackvr-9TtSrW0h-py3.11/lib/python3.11/site-packages/PyInstaller/depend/bindepend.py", line 1018, in get_python_library_path
    raise IOError(msg)
OSError: Python library not found: libpython3.11m.so, libpython3.11m.so.1.0, libpython3.11mu.so.1.0, libpython3.11.so.1.0, libpython3.11.so
    This means your Python installation does not come with proper shared library files.
    This usually happens due to missing development package, or unsuitable build parameters of the Python installation.

    * On Debian/Ubuntu, you need to install Python development packages:
      * apt-get install python3-dev
      * apt-get install python-dev
    * If you are building Python by yourself, rebuild with `--enable-shared` (or, `--enable-framework` on macOS).

make: *** [Makefile:10: pyinstaller] Error 1
```

rain-1, 2024-07-03:
The following change enabled make pyinstaller to complete:

```
PS C:\Users\Johns\Code\eyetracking\EyeTrackVR> git diff
diff --git a/Dockerfile b/Dockerfile
index e9b779b..925ed49 100644
--- a/Dockerfile
+++ b/Dockerfile
@@ -22,6 +22,8 @@ RUN  apt-get update &&\
      apt-get install python3-pip -y --no-install-recommends &&\
      pip3 install poetry
 
+RUN apt-get install python3-dev -y
+
 # Creates a non-root user with an explicit UID and adds permission to access the /app folder
 # For more info, please refer to https://aka.ms/vscode-docker-python-configure-containers
 RUN adduser -u 5678 --disabled-password --gecos "" appuser && chown -R appuser /app
 ```

rain-1, 2024-07-03:
We also require tk/tkinter:

```
diff --git a/pyproject.toml b/pyproject.toml
index 013958e..98cb204 100644
--- a/pyproject.toml
+++ b/pyproject.toml
@@ -25,6 +25,7 @@ colorama = "^0.4.6"
 taskipy = "^1.10.4"
 pytest = "^8.0.0"
 pytest-cov = "^4.1.0"
+tk = "^0.1.0"

 [tool.poetry.group.dev.dependencies]
 black = "^22.10.0"
```

```
diff --git a/pyproject.toml b/pyproject.toml
index 013958e..98cb204 100644
--- a/pyproject.toml
+++ b/pyproject.toml
@@ -25,6 +25,7 @@ colorama = "^0.4.6"
 taskipy = "^1.10.4"
 pytest = "^8.0.0"
 pytest-cov = "^4.1.0"
+tk = "^0.1.0"

 [tool.poetry.group.dev.dependencies]
 black = "^22.10.0"
 ```

but xforwarding through docker is not working on my system. so at this point I realize I will need to build this outside of docker.

RedHawk989, 2024-07-04:
Yeah, I have no experience building the app in Docker, and that is not what I would recommend. It is not an officially supported build method.

RedHawk989, 2024-07-16:
Im going to close this as it is stale and not a supported build method.