EyeTrackVR/EyeTrackVR #116: Improve Tracking/Lack thereof with threshold

C9Glax, 2024-07-12:
https://github.com/user-attachments/assets/fb60e2c5-768a-4c67-8291-5308c978b5d0

(i know that leap isn't the algorithm you want to use here...)

Don't know how well this would play with the current tracking algorithms, and whether some of these already do this.
But it would also probably make sense if there is no eye/pupil detected, that the other simulates a rough value.

RedHawk989, 2024-07-16:
RANSAC is an automatic adjusting threshold, AHSRAC and HSRAC use RANSAC for second-stage fine-tuning, and BLOB also uses a threshold.

> make sense if there is no eye/pupil detected, that the other simulates a rough value.
that is what outer eye falloff is for.

C9Glax, 2024-07-17:
The Outer Eye Falloff I see now, thought that it would only use dominant eye in that case...

Is there a possibility to manually adjust the thresholds? Every setup is slightly different, and while I am sure that these are already doing a pretty good job, I could see a human filtering out pupils/corneas even better?

RedHawk989, 2024-07-31:
All of these features are supported in the app already. I suggest you mess around with them. in the v2 app, most settings have tooltips.

closed 2024-07-31 by RedHawk989